nmunnich/apiaster #2: Chante to include an iso enter

CultVonnegut, 2025-03-06:
Hello,
I saw your post on reddit and it is really awesome and promising project, I mean I have been looking to create a keyboard for a long time, and your guide is really a comprehensive one.
I am truly surprised by the way you create the case, with a python script!? It is incredible I need to dig a little bit more on how it works.
So I wanted to ask if it would be possible to add an option to the script to add a big ass iso enter to the case, or if you have considered it and rejected it for some reason.
Thank you very much

Nick-Munnich, 2025-03-06:
Hi, that's an interesting question. I had not considered it, no. The problem is that there is no position on the PCB which could accept an ISO enter. You could technically speaking add the ISO enter to the case, but that would be a *massive* undertaking due to how I wrote the script - it uses the PCB's outline as its foundation. (Not to mention you'd get asymmetry, but it doesn't sound like that's something you'd be concerned about).

What drove your interest in ISO enter?

CultVonnegut, 2025-03-08:
Hi, thanks for response.
I also thought that maybe the pcb layout would be an inconvenient to add this feature, I also understand that it would cause an assymetry, but I am not bothered by it.
It is just a personal preference, I like pushing the big enter key, it is like this is final, go!, lol nah just nonsense. Anyways, thanks a lot for your work.
Kind regards

closed 2025-03-08 by CultVonnegut