nmunnich/apiaster #6: JLC3DP DFM warnings

nmunnich, 2025-03-13:
The 3d case has some warnings with jlc3dp dfm. I am of the belief that it should be fine unless you're using xiao, in which case you may want to expose the MCU. This issue exists to keep track of whether it has been confirmed ok yet. The outcome will then be added to the docs.

amacleod, 2025-03-27:
I am planning to make another JLC3DP order for Apiaster cases and Ripple keycaps soon. I will report here and on Discord with any notable results.

nmunnich, 2025-05-26:
It seems the warnings are unwarranted and everything is fine.

amacleod, 2025-05-26:
Forgot to follow up here, but yeah, getting a case from JLC in transparent material was just fine (though I only got the exposed MCU variant). There were one or two thin spots that I ended up breaking off, but nothing structurally significant, and not a problem if I were less clumsy handling it 🙂