nmunnich/apiaster #7: Double coin cell battery questions

Federino, 2025-03-22:
Hi! 

I am trying to build the Apiaster keyboard with 3 MCUs to make it full wireless, but I have some doubts regarding the coin cell bateries:

1) In the building guide there are only pictures for the left side soldering of the coin cell container, should the right side of the keyboard have a coin cell as well? Also, should the coin cell container on the right side mirror the left side or should both have the same "direction"? I am asking because if you follow the printed signs on the right side for the container the orientation is not mirrored.

2) If I run out of battery, I assume that I would have to plug both left and right side with each other and then plug the left side with an USB to the computer in order to charges both halves, is this correct? If this is correct, doesn't this imply that even if you want a full wireless build you also need the USB A port for both halves?

I am sorry if these are stupid questions, but its my first time building a full wireless keyboard.

Thank you for your time!

nmunnich, 2025-03-24:
Hi! Good questions:

1. The right side will also need a coin cell. The holder for this cell should match the silkscreen on the PCB, in other words they should be mirrored. I'm a bit confused what you mean by them not being mirrored, perhaps an image here would help?
2. If you run out of battery you will want to charge each half individually by connecting their respective MCUs to a source of charge, be it a charging brick or a computer. The USB-A port cannot be used to charge.

Federino, 2025-03-25:
Hi!

First of all, thank you for the answers, they were really helpful!

What i meant with the mirrored coin cell position was that at first I saw the inprint of the right side part and I thought that the + and - position were the same in both sides, meaning:[ + - ] on the left side and [+ -] on the right side, meaning that the coin cell container would have the exact same position in both sides, making it look like both containters were looking in the same direction, but I was wrong... The inprint is correct, I just soldered the right side container in the wrong polarization (?).

Also, now that I am mentioning the coin containers, I found a problem with the default STL files for the Choc version. specifically the _apiaster-l-top_coin-fle_x from both sides. With the coin containers from the buying list, the top of the coin container hits the slope that is on the coin cover, making it so that you can't close the case correctly. I fixed this sort of cutting the part of the coin container so that it is shorter and doesnt hit the slope from the case. I add the file with the slope that I am talking about. 

If you need more information or anything just let me know.

Thank you very much!

[image]

nmunnich, 2025-03-26:
Thanks for letting me know. I think the slant in that area is just lacking a bit of tolerance - my printed version seems to fit fine in some positions of the coin cell, while being a bit too loose in others. I believe I oversized the holes of the coin cell connector a bit, in case someone wanted to put a different beefy battery there instead with a completely different case.

I've updated the script and it should be improved now. I'll update the individual files as requests come in, but I believe people should be able to get it to fit by adjusting the exact position of the battery connector while soldering.

Will close this and open a new issue to track.

closed 2025-03-26 by nmunnich