nmunnich/apiaster #9: Dual batteries?

manherna, 2025-03-27:
Hi! 
First let me say thanks for the amazing project and all the work that has been put into it. 
I am currently building several apiasters with some friends and it rocks! Also docs are really easy to follow and get everything right.
Just one question. When using only one XIAO, do we need both halves with coin battery? Or only the left one?

Thanks!

nmunnich, 2025-03-27:
You only need batteries on the left side. One battery per XIAO.

manherna, 2025-03-28:
I see, thanks. 
And can I leave the connector and charge protection soldered in the right in case I want to upgrade?

nmunnich, 2025-03-28:
Yes you can, though I would recommend not actually inserting a battery until you need to make use of it.

manherna, 2025-03-28:
Oki, thanks for the info! Closing issue

closed 2025-03-28 by manherna