nmunnich/apiaster #10: Keyboard undetected

Federino, 2025-03-29:
Hi! 

I built the entire keyboard but I can't use the keyboard at all. I built the entire thing with the USB dongle functionality in mind, so after I soldered everything, I flashed the firmware into the 3 chips: for the dongle I used the `apiaster-dongle-xiao-ble-studio.uf2` file, for the right side I used the `apiaster-right-xiao-ble.uf2` file and for the left side I used the `apiaster-left-xiao-ble-with-dongle.uf2`.

After that I did a series of tests to make it work, but I could not. If I plug the halves and the USB dongle, the halves have a very light green LED turned on, but the dongle has not light at all.

If I press the RESET button from the dongle while connected with an USB I get 2 yellow led flashes and then it turns off.

If I press the RESET button of either half, I get a bright GREEN LED flash, followed by a single RED led flash and then both halves turn off.

Shouldn't I get at least the ability to type if both halves are connected (separately) to the computer with 2 USB cables? Do I need to configure something more before starting to type? Is there any configuration I have to make to the dongle in order for it to keep working?

I also tryed connecting the dongle to ZMK Studio and the dongle appears as USB Serial Device (COM9) and after trying to connect to it I get "Unlock to continue" error. Didn't the dongle firmware come with it unlocked already?

All the software I used is in the `firmware.zip` from this repo.

I hope I can help!

Thank you very much!

nmunnich, 2025-03-29:
Hi, if you flashed the dongle setups then you can only use the keyboard when the dongle is connected. ZMK Studio is *enabled*, but for security purposes when you wish to use it you need to press the `&studio_unlock` behavior in the keymap. The keymap can be found in the README of the firmware repo. 

Regarding the light flashing, the firmware from the repo makes use of https://github.com/caksoylar/zmk-rgbled-widget?tab=readme-ov-file#features, which you can look at to see what the light flashes mean. The first flash is battery, the second is connection status. It sounds like your halves aren't connecting to the dongle properly: make sure they're both powered on, then press the reset button on both simultaneously. See also https://zmk.dev/docs/troubleshooting/connection-issues#split-keyboard-halves-unable-to-pair, and the rest of the troubleshooting section of the ZMK docs.

Let me know how you get on.

Federino, 2025-03-30:
Hi! I have managed to reflash the software on the left and right side to make them work with the left side as the central part and the right side as slave. I have not managed to make them run with the dongle.

I tryed to turn the dongle and both halves at the same time, but they do not connect to it. This week i'll try and see if there is something more I can do. I'll keep you updated if I manage to fix it. 

Thank you!

nmunnich, 2025-03-30:
Hi, I just doublechecked and there was indeed an issue with the dongle firmware - I had forgotten to push a local commit. Give the newly built firmware a try, please.

Federino, 2025-03-30:
Hi, I just tested it and still does not work. I flashed the reset on all 3 chips, and then uploaded the following files: ``apiaster-dongle-xiao-ble-studio.uf2``, ``apiaster-left-xiao-ble-with-dongle.uf2`` and ``apiaster-right-xiao-ble.uf2``. 

After that, I pressed the ``reset`` button in all three components and they do not connect. The dongle flashes a green light, followed by a yellow one and then turns off. The keyboard halves both turn on with a green LED, followed by a red flash and then both turn off.

The dongle appears as Apiaster in the bluetooth devices, but the halves won't connect. 

If you need anything else let me know.

Thank you very much!

Federino, 2025-03-30:
Okay, I don't exactly know that happened, but I uploaded the ``reset`` files on both halves, reflashed both of them with their respective firmwares and they connected to the dongle. I would love to provide further help, but I have no clue what changed hahahaa

nmunnich, 2025-03-30:
If it works it works! One more thing: Would you perhaps want to share some images of your finished keyboard? I'm collecting pictures so that I can add a gallery section to the repo.

Federino, 2025-03-30:
Sure! But it might take some time... I think one of the chips just died somehow...

nmunnich, 2025-03-30:
The likeliest sources of damage would be:
1. Faulty batteries
2. ESD discharge when handling components (note that this can happen even if the device is in its case, it is just quite a bit rarer)
3. Plugging in the USB-A 3.0 cable between halves while both controllers are powered *might* cause some damage
4. short circuiting the plug on the xiao somehow

Federino, 2025-03-30:
If the green LED is not turning on even if the chip is not in the PCB should I asume it is completely dead? It died after I tryed to install a new firmware with a different layout. Could it be that I did not generate the version correctly?

nmunnich, 2025-03-30:
If you used my firmware repo as a module and then only edited the keymap, and generated firmware for the correct board, there should be no reason whatsoever for your firmware to break the chip. The green LED signifies the charging state of the device, if there is no battery connected (or the battery is full) it should not light up. 

I would try with alternative cables/usb port just to be sure, but if you're unable to enter the bootloader regardless or what you try then my hunch would be an ESD event occurred. You can avoid future ESD events when handling electronics with your bare hands by regularly discharging yourself e.g. by touching the metal case of a plugged in desktop or a grounded metal radiator.

nmunnich, 2025-03-30:
Ah, just to be certain we're on the same page: Bootloader should be able to be entered by pressing the teeny tiny button on the left of the usb port twice within 500ms.

Federino, 2025-04-06:
Hi! Sorry for beeing so late to reply, but I think the chip was indeed fried. I think it had to do with me using a magnetic usb for powering it, but I can't be sure. I was not able to enter Bootloader.

As for this issue, with the fix you provided last week, everything worked perfectly with the dongle (until I burned it), so thank you very much! I ended up having to flash the version making the left side as the central device and everything worked flawlessly.

Also, as you requested, here are some screenshots of the final build:

[image]
[image]

If you need to try something else or debug something I can help, please let me know!

Thank you for your time and this amazing proyect!

closed 2025-04-06 by Federino