nmunnich/apiaster #12: Create a case with adjusted thumb cluster location

Etgreg, 2025-04-04:
I was wondering if you could generate a case for my particular use. I would need a thumb adjustment of 0,20,5.

Thanks

nmunnich, 2025-04-05:
Could you please specify all the other settings you want? e.g. MCU, battery, inner key column, magsafe, etc. etc. etc.

Etgreg, 2025-04-07:
MX, either MCU, Lipo battery, expose MCU, all inner column

Everything else should be default. Thanks

nmunnich, 2025-04-11:
Hi, unfortunately it seems like life has become rather busy for me lately. I unfortunately do not know when I can get around to this - it would probably be significantly quicker for you to generate them yourself.

nmunnich, 2025-05-26:
I've moved to using GHA for case generation. This should make it much more feasible for anyone to generate the case for themselves. As such, I'm going to close this. Please try following the new GHA instructions, and reopen if you are struggling.

nmunnich, 2025-06-15:
Considering only the left half for simplicity of terminology:

The origin of the thumb adjustments (both for rotation and movement) is at the top left corner of the leftmost thumb. As such, *any* negative x/y movement values will immediately cause collision, so I have banned those. The only exception would be if you have a y offset of at least 19mm, but then the case generator will bug out.

On a more motivational note: I am of the strong belief that putting your thumb in line with your index is the tuckiest it should be for long-term thumb health. The only person I know for whom tuckier is desirable has seriously damaged thumbs due to a long history of fighting games with un-ergonomic controllers. The furthest inwards the thumbs go on Apiaster is 3.5mm further to the right of the index.