nmunnich/apiaster #17: Share the original case files in a different format?

togden, 2025-05-30:
Hi, I'm interested in building an Apiaster, but want to modify the case a bit. The STLs that get generated aren't very easy to modify, and I was wonering if you'd be willing to share the files in a different file format, like STEP?

Cheers!

nmunnich, 2025-05-30:
The entire case is written using build123d from the ground up. If you want a step file, you can edit the code in your fork to export a step instead of an stl. Should be a very simple find and replace task. That said, I have no idea how helpful a build123d generated step file would be. Would be curious if you decided to report back.

https://build123d.readthedocs.io/en/latest/import_export.html

togden, 2025-06-02:
Did a basic find & replace for stl -> step and with a bit of troubleshooting got the code to run, but there was no output. Will require a bit more in-depth work to get the export to work, it seems. Will keep chipping away at it.